BiancoRoyal/node-red-contrib-iiot-opcua #99: How-To OPC UA Server setup

freaki2010, 2019-06-06:
**I'm submitting a ...**  (check one with "x")
## Support
 - support request => **Please do not submit support requests here**, instead see use [the link to our support page](https://bianco-royal.cloud/supporter/#/content/backer)
 - (x) simple question => please visit the Wiki and Info tab by clicking on the node inside Node-RED before
Where can I find a detailed How-To OPC UA Server setup? Nothing is written under the tab [Server](https://github.com/BiancoRoyal/node-red-contrib-iiot-opcua/wiki/Server) in the [Wiki](https://github.com/BiancoRoyal/node-red-contrib-iiot-opcua/wiki) and the details in Node-Red itself don't really help me. I am a trainee of a company and got the task to run an OPC-UA server with Node-Red on the IOT2040. 
Thanks in advance for the help!

biancode, 2019-06-06:
There is no detailed How-To OPC UA Server setup. If you write one, then let us know and we add it to the Wiki.

closed 2019-06-06 by biancode

freaki2010, 2019-06-06:
What you own cannot be called a wiki.
I would suggest that you take a look at some other wikis and revise yours.
You can't use your wiki for information purposes.
That's not meant to be bad, it's just a fact. sorry...

biancode, 2019-06-06:
sure - I never said something else - feel free to contribute instead to diss my work, maybe with that, you will see how much work it is to provide more ...

biancode, 2019-06-06:
Sure, feel free to make it a better one!

biancode, 2019-06-06:
[image]

biancode, 2019-06-06:
[Wiki Upgrade project ](https://github.com/BiancoRoyal/node-red-contrib-iiot-opcua/projects/6) needs more contributors