BiancoRoyal/node-red-contrib-iiot-opcua #100: Unable to write arrays (v3)

mm-dlwe, 2019-06-07:
**I'm submitting a ...** 

## Problem
 - (x) bug report (might be related to #51, it does not work in v3 even if it seems like it should have been fixed)

### Information
 **Current behaviour**

 Writing an array produces an invalid message where only a value of 0 is send instead of the array. For this reason the OPC server responds with an error.

 Response from the OPC server:
 - description: "The value supplied for the attribute is not of the same type as the attribute's value."
 - name: "BadTypeMismatch"

  Reason for this behaviour is the following code sequence:
   1. opcua-iiot-write.js: node.on('input', ...) is called
   2. opcua-iiot-write.js: node.bianco.iiot.writeToSession is called to initiate the write request.
   3. opcua-iiot-write.js: coreClient.core.buildNodesToWrite is called to create the request
   4. core/opcua-iiot-core.js: de.biancoroyal.opcua.iiot.core.buildNodesToWrite tries to create a request for each listed node and calls de.biancoroyal.opcua.iiot.core.buildNewVariant
   5. core/opcua-iiot-core.js: de.biancoroyal.opcua.iiot.core.buildNewVariant tries to convert the value type and does not respect arrayType, it tries to parse the array as integer with results in a value of zero instead of an array
   6. The message is send to the OPC-UA Server which responds with a BadTypeMismatch failure because it expects an array and not a value.
 
 **Expected behaviour**
  Instead of a single value the whole array should be passed to the node-opcua library and the OPC server. The write operation should then suceed.

 **Minimal reproduction of the problem with instructions**
   1. Setup a flow and try to write an array (see the attached demo flow)

 **Minimal Demo Flow**
 [flows.txt](https://github.com/BiancoRoyal/node-red-contrib-iiot-opcua/files/3264938/flows.txt)
   
#### Your Environment
**Please tell us about your environment:**
  * (x) I am using just the Node-RED package and got node-opcua installed with it
  * Device :
    - (x) Windows: 10
  * Description of the connecting OPC UA system:
     - Name   : UaCPPServer
     - Version: 1.5.6
     - Vendor : Unified Automation
     - Link   : https://www.unified-automation.com/downloads/opc-ua-servers.html
     
* **Node-RED version:** 0.20.5
* **node-red-contrib-iiot-opcua version:** 3.10.2 (a0c60da)
* **Node:** (min. 8.x)
   `node --version` = v8.11.3

biancode, 2019-06-08:
The full array handling is not implemented for now.

mm-dlwe, 2019-07-02:
I#ve added a possible solution in PR #109.

biancode, 2019-11-26:
feat since prerelease 3.11.1-4 - Matrix and Array is working well on testing

biancode, 2019-12-03:
released with 3.11.1

closed 2019-12-03 by biancode