BiancoRoyal/node-red-contrib-iiot-opcua #102: Sampling Interval = 0

MileBuurmeijer, 2019-06-16:
**I'm submitting a bug report 
   
## Problem
 - ( ) bug report => please search GitHub for a similar issue or PR before submitting
 - (*) unexpected behaviour => please visit the Wiki before

### Information
 **Current behavior**
When subscribing to an OPC UA server with 2 monitored items in a subscription and each with sampling interval set to zero, not every data value the OPC UA server creates is sent to this Node-Red IIOT OPC UA client
 
 **Expected behavior**
 When setting sampling interval to zero on a monitored item the expected behavior is that each data value the OPC UA server creates is sent to this Node-Red IIOT  OPC UA client
 
 **Minimal reproduction of the problem with instructions**
 
   1. Created an OPC UA Server that simulates a sine and a cosine signal with a sampling rate of 5000 samples per second ( a Milo OPC UA server based project)
   2. Checked with UA Expert that it generates that much data per second with the UA Expert datalogger feature
   3. Created in Node-Red the following Node OPCUA Based flow (I hope interval on the inject node means sampling interval and if so it would be ok to rename interval to sampling-interval)

```javascript
[{"id":"eaa0d412.4cd698","type":"OPCUA-IIoT-Inject","z":"cb7e86be.302b18","injectType":"listen","payload":"{\"interval\":0,\"queueSize\":1000,\"options\":{\"requestedPublishingInterval\":500,\"requestedLifetimeCount\":60,\"requestedMaxKeepAliveCount\":10,\"maxNotificationsPerPublish\":500,\"publishingEnabled\":true,\"priority\":1}}","payloadType":"json","topic":"","repeat":"","crontab":"","once":true,"startDelay":"10","name":"Start/stop listening","addressSpaceItems":[{"name":"RunState","nodeId":"ns=2;s=Player/Runstate","datatypeName":""},{"name":"Sinus","nodeId":"ns=2;s=1001","datatypeName":""},{"name":"Cosinus","nodeId":"ns=2;s=1002","datatypeName":""}],"x":145,"y":178.99998259544373,"wires":[["e18fdb5d.95c838"]]}]
```   

#### Your Environment
**Please tell us about your environment:**

  * ( *) I am using just the Node-RED package and got node-opcua installed with it
  * ( ) I have installed node-opcua global from source (using git clone)
  * ( ) I have installed node-opcua as a global package (using npm i -g) 
  
  * Device :
    
    - ( ) macOS: v 
    - (x ) Linux: v 19.04 (Unbuntu)
    - ( ) Windows: v 
    - ( ) Raspbian: v 
    - ( ) Other:
         - Name:
         - Version:
   
   * Description of the connecting OPC UA system:
   
     - Name   : OPC UA Milo server
     - Version: 0.2.1
     - Vendor : Ecplise foundation - Milo OPC UA project
     - Link   : https://github.com/eclipse/milo
     
* **Node-RED version:** 
0.20.5

* **node-red-contrib-iiot-opcua version:** ``
3.10.20

* **Node:** (min. 8.x)
   `node --version` = 9.10.1

MileBuurmeijer, 2019-08-03:
Finally I had the time to do testing of the underlying node-opcua node-module. I edited the sample-client.ts in the node-opcua-samples package so that it sets the sampling interval of the monitored item to zero. When I ran this sample-client.ts is published far more values than with the original sampling interval of 1000 in the sample client. I have no means to check if the used OPC UA server (opc.tcp://milo.digitalpetri.com:62541/milo) has sent all the samples that occurred, but for now, I think I can conclude the node-opcua package does support setting sampling interval to zero and that node-red-contrib-iiot-opcua should be able to support setting sampling interval to zero. If only I was more fluent in javascript I could check where in the source code it checks for non-zero values of the sampling interval. Could someone point me in the right direction? If so I would like to try to change it myself and do some more testing.

MileBuurmeijer, 2019-08-06:
Did some more testing and found out that node-js itself is very fast. I could receive up to 7000 samples per second from my OPC UA server. The server was derived from the Eclipse Milo project (see github [OPCUA-Player](https://github.com/MileBuurmeijer/OPCUA-Player). Any ideas if node-red would be capable of processing up to 7000 s/s?

biancode, 2019-12-14:
Minimum is set to 0 now -- you can test it if it works for you and if node-opcua will accept it

closed 2019-12-14 by biancode