BiancoRoyal/node-red-contrib-iiot-opcua #104: Not able to get it working within node-red docker

bsense-rius, 2019-06-19:
**I'm submitting a ...**  (check one with "x")
   
## Feature
 - ( ) feature request
 
 **What is the motivation or Use Case for your feature request?**
 
 
 
I am submitting this issue as I have not been able to get these nodes working inside node-red dcoker container. I have tested it with local installations (windows and linux) with much success. But no matter how I tried it (with persistence, without etc) the ASO server is not running. It also does not report any error when running (in nodered command line.

## Support
 - support request => **Please do not submit support requests here**, instead see use [the link to our support page](https://bianco-royal.cloud/supporter/#/content/backer)
 - ( ) simple question => please visit the Wiki and Info tab by clicking on the node inside Node-RED before
 - FAQ: Can I send you individual support requests via E-Mail? 
   - Yes, sure! :-D
   
   
   

## Problem
 - ( ) bug report => please search GitHub for a similar issue or PR before submitting
 - ( ) unexpected behaviour => please visit the Wiki before

### Information
 **Current behaviour**
 
 
 **Expected behaviour**
 
 
 **Minimal reproduction of the problem with instructions**
 
   1. 
   2. 
   3. 
   

#### Your Environment
**Please tell us about your environment:**

  * ( ) I am using just the Node-RED package and got node-opcua installed with it
  * ( ) I have installed node-opcua global from source (using git clone)
  * ( ) I have installed node-opcua as a global package (using npm i -g) 
  
  * Device :
    
    - ( ) macOS: v 
    - ( ) Linux: v 
    - ( ) Windows: v 
    - ( ) Raspbian: v 
    - ( ) Other:
         - Name:
         - Version:
   
   * Description of the connecting OPC UA system:
   
     - Name   :
     - Version:
     - Vendor :
     - Link   : https://
     
* **Node-RED version:** 

* **node-red-contrib-iiot-opcua version:** ``

* **Node:** (min. 8.x)
   `node --version` =

biancode, 2019-06-19:
reopen with more details please, mostly which versions you are using there

closed 2019-06-19 by biancode

bsense-rius, 2019-06-19:
I am very sorry! My fault. It works also like charm. I missed to map also the OPCUA port in the docker run command (-p 1880:1880 -p 55388:55388)

biancode, 2019-06-20:
No problem. Please, give us more details about versions in your next issue and we are happy to get your issue.