BiancoRoyal/node-red-contrib-iiot-opcua #105: Inject Node shows no configuration options.

youngkzy, 2019-06-19:
I filled in the blanks below and described the issue.

**I'm submitting a ...**  (check one with "x")
   
## Feature
 - ( ) feature request
 
 **What is the motivation or Use Case for your feature request?**
 
 
 

## Support
 - support request => **Please do not submit support requests here**, instead see use [the link to our support page](https://bianco-royal.cloud/supporter/#/content/backer)
 - ( ) simple question => please visit the Wiki and Info tab by clicking on the node inside Node-RED before
 - FAQ: Can I send you individual support requests via E-Mail? 
   - Yes, sure! :-D
   
   
   

## Problem
 - (X) bug report => please search GitHub for a similar issue or PR before submitting
 - ( ) unexpected behaviour => please visit the Wiki before

### Information
 **Current behaviour**
 
Nothing inside the configuration on OPC-IIot Inject node.
 
 **Expected behaviour**
 
 
 **Minimal reproduction of the problem with instructions**
 
   1. Clicked on the OPC IIoT Inject
   2. Nothing is inside it.
   3. 
   

#### Your Environment
**Please tell us about your environment:**

OPC Connections seem to work OK.

  * ( ) I am using just the Node-RED package and got node-opcua installed with it
  * ( ) I have installed node-opcua global from source (using git clone)
  * (X ) I have installed node-opcua as a global package (using npm i -g) 
  
  * Device :
    
    - ( ) macOS: v 
    - ( ) Linux: v 
    - (X ) Windows: v    Windows_NT 10.0.16299 x64 LE
    - ( ) Raspbian: v 
    - ( ) Other:
         - Name:
         - Version:
   
   * Description of the connecting OPC UA system:
   
     - Name   : Kepware KepServerEX5 Allen Bradley Suite
     - Version: 5
     - Vendor : Kepware
     - Link   : https://
     
* **Node-RED version:** 
 Node-RED version: v1.0.0-beta.2

* **node-red-contrib-iiot-opcua version:** ``

* **Node:** (min. 8.x)
   `node --version` =   Node.js  version: v10.15.1

PS C:\Users\me.FOXCONN-NA> node-red
19 Jun 11:39:48 - [info]

Welcome to Node-RED
===================

19 Jun 11:39:48 - [info] Node-RED version: v1.0.0-beta.2
19 Jun 11:39:48 - [info] Node.js  version: v10.15.1
19 Jun 11:39:48 - [info] Windows_NT 10.0.16299 x64 LE
19 Jun 11:39:49 - [info] Loading palette nodes
19 Jun 11:39:55 - [info] Dashboard version 2.15.4 started at /ui
19 Jun 11:39:56 - [info] Settings file  : \Users\youngk.FOXCONN-NA\.node-red\settings.js
19 Jun 11:39:56 - [info] Context store  : 'default' [module=memory]
19 Jun 11:39:56 - [info] User directory : \Users\youngk.FOXCONN-NA\.node-red
19 Jun 11:39:56 - [warn] Projects disabled : editorTheme.projects.enabled=false
19 Jun 11:39:56 - [info] Flows file     : \Users\youngk.FOXCONN-NA\.node-red\flows_APCCYOUNGKL.json
19 Jun 11:39:56 - [info] Server now running at http://127.0.0.1:1880/
19 Jun 11:39:56 - [warn]

scelle, 2019-08-30:
Same problem here.
Extra info
When loading the Node-Red UI page, I get the following error message in Chrome:

VM504:1 Uncaught SyntaxError: Identifier 'injectNode' has already been declared
    at <anonymous>:1:1
    at b (vendor.js:2)
    at Ie (vendor.js:2)
    at e.fn.init.append (vendor.js:2)
    at r (red.min.js:16)
    at i (red.min.js:16)
    at r (red.min.js:16)
    at i (red.min.js:16)
    at r (red.min.js:16)
    at i (red.min.js:16)

Workaround for now: export the node, edit json, import again

kind regards
Pascal

piyush-gupt, 2019-10-08:
I faced the same problem. Although it was working perfectly fine in version 0.7 but as soon as I updated to version 1.0.1, the edit configuration for OPC items was missing.

skaugify, 2019-10-31:
I'm also having this problem on a Wago PFC200 8212 PLC.
Here's the version: https://hub.docker.com/r/wagoautomation/node-red-iot

Rainotronik, 2019-11-18:
I have the same problem on a WAGO PFC200 8212. With both: FW12 and FW14

scorbin-ewi, 2019-11-26:
same issue.  

Node Red version: 1.0.1
node-red-contrib-iiot-opcua version:  3.10.2
KepServer 6 with Mitsubishi and Modbus Suite
Debian Buster (10)  all mainstream updates current

Worked prior to Node Red 1.0.1. Now only way to edit Inject nodes is to export json, edit, and import json again as scelle suggested.  However I do not get dashboard error that scelle mentioned.

biancode, 2019-11-26:
you could test prerelease 3.11.1-4 - button in Node-RED changed with 1.x so it is a duplicate of #106

closed 2019-11-26 by biancode

scelle, 2019-11-27:
Ok, I replaced opcua-iiot-inject.html with the version of the prerelease.  This works fine now.
Thanks @lemariva @biancode