BiancoRoyal/node-red-contrib-iiot-opcua #106: Node-RED v1.x and NodeJS 12

biancode, 2019-06-20:
**I'm submitting a ...**  (check one with "x")
   
## Feature
 - (X ) feature request
 
 **What is the motivation or Use Case for your feature request?**
 
 
 
     
* **Node-RED version:** 
v1.x

* **node-red-contrib-iiot-opcua version:** ``

* **Node:** (min. 8.x)
   `node --version` = 12.x

biancode, 2019-10-03:
I stopped contributing to Node-RED for now. I have to write my master thesis first.

biancode, 2019-10-05:
Sponsors and interested collaborators are welcome https://contact.bianco-royal.com/

biancode, 2019-12-03:
Node-RED v1.x with v3.11.1+

Tom-Simon, 2020-02-03:
Hello,

is this still the case? I've updated NodeJS 12.4.1 with Node-Red 1.0.3 and can use your great opc ua nodes without any new problems?

biancode, 2020-02-03:
@Tom-Simon, for now, I just tested with Nodejs 10

closed 2022-05-27 by biancode