BiancoRoyal/node-red-contrib-iiot-opcua #107: Flex Connector does not connect to OPCUA gateway and causes crash

Pasc95, 2019-06-25:
## Problem
 - ( ) bug report => please search GitHub for a similar issue or PR before submitting
 - ( x ) unexpected behaviour => please visit the Wiki before

### Information
 **Current behaviour**
 
I am using the new Flex Connector with the following payload to dynamically use the IP address of a specific OPCUA gateway:

[image]

But after triggering the connection the Node-red application always crashes with the following error message: 

```
TypeError: Cannot read property 'toString' of undefined
    at /Users/de164582/.node-red/node_modules/node-opcua-client/src/client_base.js:246:51
    at /Users/de164582/.node-red/node_modules/node-opcua-client/node_modules/async/dist/async.js:3888:9
    at /Users/de164582/.node-red/node_modules/node-opcua-client/node_modules/async/dist/async.js:473:16
    at replenish (/Users/de164582/.node-red/node_modules/node-opcua-client/node_modules/async/dist/async.js:1006:25)
    at iterateeCallback (/Users/de164582/.node-red/node_modules/node-opcua-client/node_modules/async/dist/async.js:995:17)
    at /Users/de164582/.node-red/node_modules/node-opcua-client/node_modules/async/dist/async.js:969:16
    at Immediate._onImmediate (/Users/de164582/.node-red/node_modules/node-opcua-client/node_modules/async/dist/async.js:3885:13)
    at runCallback (timers.js:810:20)
    at tryOnImmediate (timers.js:768:5)
    at processImmediate [as _immediateCallback] (timers.js:745:5)
```
 
I also tried to put the parameters into the msg object (not payload) because it is not precisely documented in the node-red node description. But it didn't work neither.

 **Expected behaviour**
I expect the gateway to be connected after the Flex Connector node received the payload shown above.
 
 **Minimal reproduction of the problem with instructions**
 
   1. Flex Connector
   2. Fill the payload with a function and the following content: 
```
msg.payload = {}
msg.payload.endpoint = 'opc.tcp://9.241.158.49:4129/'
msg.payload.keepSessionAlive = true
msg.payload.securityPolicy = 'None'
msg.payload.securityMode = 'None'
msg.payload.name = 'bms129lgwp01'
msg.payload.endpointMustExist = false
msg.payload.autoSelectRightEndpoint = false  
msg.payload.showErrors = true
msg.payload.publicCertificateFile = null
msg.payload.privateKeyFile = null
msg.payload.discoveryUrl = null
msg.payload.defaultSecureTokenLifetime = 0
msg.payload.strategyMaxRetry = 0
msg.payload.strategyInitialDelay = 0
msg.payload.strategyMaxDelay = 0
msg.payload.strategyRandomisationFactor = 0
msg.payload.requestedSessionTimeout = 0
msg.payload.connectionStartDelay = 0
msg.payload.reconnectionDelay = 0
return msg
```
   3. Inject
   

#### Your Environment
**Please tell us about your environment:**

  
  * Device :
    
    - ( x ) macOS: v10.14.5
    - ( ) Linux: v 
    - ( ) Windows: v 
    - ( ) Raspbian: v 
    - ( ) Other:
         - Name:
         - Version:
     
* **Node-RED version:** 
v0.19.4

* **node-red-contrib-iiot-opcua version:** ``
v13.10.2

* **Node:** (min. 8.x)
   `node --version` = v8.15.0

biancode, 2019-11-26:
please test with prerelease 3.11.1-4

biancode, 2019-12-14:
reopen if it is also in v3.11.2 or above

closed 2019-12-14 by biancode