BiancoRoyal/node-red-contrib-iiot-opcua #108: When writing '0' as string to a Boolean node sets it to true

mm-dlwe, 2019-07-02:
**I'm submitting a ...**  (check one with "x")
   

## Problem
 - (x) bug report

### Information
 **Current behaviour**
When writing a '0' as string to a Boolean node the value is set to true instead of false.
The test 'should build new variant Boolean' in opcua-iiot-core.test.js looks like using 1 or 0 should actually be supported. But the test checks for '1' only and it is '0' that does not work.

A fix would probably be to change the condition in opcua-iiot-core.js at line 243
from
`if(value && value !== 'false')`
to
`if(value && value !== 'false' && value !== '0')`
 
 **Expected behaviour**
Using '0' should be interpreted as false.
 
 **Minimal reproduction of the problem with instructions**
   1. Setup a flow and try to write a boolean using a '0' string (see the attached demo flow)

**Minimal Demo Flow**
[flows-boolean.txt](https://github.com/BiancoRoyal/node-red-contrib-iiot-opcua/files/3349116/flows-boolean.txt)

**Your Environment**

**Please tell us about your environment:**

   * (x) I am using just the Node-RED package and got node-opcua installed with it

   * Device :
       * (x) Windows: 10

   * Description of the connecting OPC UA system:
       * Name : UaCPPServer
       * Version: 1.5.6
       * Vendor : Unified Automation
       * Link : https://www.unified-automation.com/downloads/opc-ua-servers.html
   
* **Node-RED version:**  0.20.5

* **node-red-contrib-iiot-opcua version:** 3.10.2 (a0c60da)

* **Node:** (min. 8.x)
   `node --version` = v8.11.3

biancode, 2019-07-02:
There was a reason, why I did not check for '0' to be used as a false ... I'll check it.

biancode, 2019-11-26:
fixed since prerelease 3.11.1-4

closed 2019-11-26 by biancode