BiancoRoyal/node-red-contrib-iiot-opcua #111: Cannot connect to OPCUA Demo Server from Unified Automation (UaNetServer) using Node-RED IIoT Read or Write

parth-ifam, 2019-09-17:
### Issue
Unable to connect to OPCUA Server created by Unified Automation .NET classes

### What happens?
- The IIoT OPCUA Read or Write block shows connecting but cannot connect to the Server.
- Server is the demo server (ServerGettingStarted_Net40.sln) from Unified Automation .NET classes. Example Lesson01
- No errors are reported by Node-RED except Error Stopping the Nodes_ Close timed out _upon deploying_
- Both server and client are running locally. With UaExpert it is possible to connect to the server
- The server does not have any security and certificates are not used.
- OPCUA Read and Write can connect to other OPCUA servers not programmed with Unified Automation.

### Please tell us about your environment:

- [x] Node-RED version: 0.20.8
- [x] Node.js version: 10.16.3
- [x] npm version: 6.9.0
- [x] Platform/OS: Windows 10
- [x] Browser: Mozilla Firefox 60.7.2esr

biancode, 2019-09-17:
Hi, which version of the UA-SDK is it, please?

parth-ifam, 2019-09-18:
Hello, it is UA .NET SDK version 3.0.0. Please see the link below. It already has some examples which i listed in the previos message.
https://www.unified-automation.com/products/server-sdk/net-ua-server-sdk.html

biancode, 2019-12-14:
[image]
[image]
[image]

working well ... maybe you did not trust the certs with moving the cert file on UA server side

closed 2019-12-14 by biancode