BiancoRoyal/node-red-contrib-iiot-opcua #112: Support for custom Data Types

skarux, 2019-09-17:
## Feature
 - (x) feature request
 
 **What is the motivation or Use Case for your feature request?**
 
 
 Currently only default OPC UA data types are supported; typing a custom data type in the "OPCUA-IIOT-Node" node the final value attribute gets missing or set to null (depending on the chosen options) and on the following error is displayed on the console:
`#getConstructor : cannot find constructor for expandedId  ns=2;i=6`
is reported by the inner OPC UA library. It would be nice to have at least a byte array containing the raw value to be parsed with some external functions.

 

#### Your Environment
**Please tell us about your environment:**

  * ( ) I am using just the Node-RED package and got node-opcua installed with it
  * ( ) I have installed node-opcua global from source (using git clone)
  * (x) I have installed node-opcua as a global package (using npm i -g) 
  
  * Device :
    
    - ( ) macOS: v 
    - (x) Linux: Ubuntu 18.04 
    - ( ) Windows: v 
    - ( ) Raspbian: v 
    - ( ) Other:
         - Name:
         - Version:
   
     
* **Node-RED version:  v0.20.8** 

* **node-red-contrib-iiot-opcua version:** `v3.10.2`

* **Node:** (min. 8.x)
   `node --version` = 8.16.1

piyush-gupt, 2019-10-09:
I have experienced the same problem where I want to get an array of user defined structures but getting null. Moreover the problem lies not only with OPC UA IIOT contrib but normal OPC UA contrib also.

biancode, 2019-12-14:
@erossignon do you have an idea here with v0.7.4 of node-opcua or is it just to do with v2.x?

erossignon, 2019-12-15:
@biancode this will need v2

closed 2022-05-27 by biancode