BiancoRoyal/node-red-contrib-iiot-opcua #113: Injecting 0 for any integer node is not working for Demo Server

prrw, 2019-09-18:
## Problem
 - (x) bug report => Injecting 0 for any integer node is not working

### Information
 **Current behaviour**
 
In the standard example "**_IIoT OPC UA ASO Types_**", if we inject 0 to UInt64, Int64, Int32, Int16, Double or Float, it does not work.

- Int16 and Int32 throw an error: ```Error: Expecting a number```
- Int64 and UInt64 throw an error. ```Error: Error```
- Double or Float do not throw any error but it writes 'nan' to the server
 
#### Your Environment
**Please tell us about your environment:**

  * ( ) I am using just the Node-RED package and got node-opcua installed with it
  * ( ) I have installed node-opcua global from source (using git clone)
  * (x) I have installed node-opcua as a global package (using npm i -g) 
  
  * Device :
    
    - ( ) macOS: v 
    - ( ) Linux: v 
    - (x) Windows: Windows 10 
    - ( ) Raspbian: v 
    - ( ) Other:
         - Name:
         - Version:
   
* **Node-RED version:**  `0.20.8`

* **node-red-contrib-iiot-opcua version:** `3.10.2`

* **Node:** (min. 8.x)
   `node --version` = `10.16.3`

biancode, 2019-09-18:
Hi, provide an example flow, please!

prrw, 2019-09-18:
[flows.zip](https://github.com/BiancoRoyal/node-red-contrib-iiot-opcua/files/3627048/flows.zip)
[image]

biancode, 2019-12-14:
working with v5.2.0
[image]
[image]

biancode, 2019-12-14:
UInt64 does not work anymore

biancode, 2019-12-14:
Double gets nan on 0.0 init

biancode, 2019-12-14:
convert must do a Number() wrap for node-opcua 0.7.4
[image]

biancode, 2019-12-14:
Example flow
[ZeroASOFlowExample.json.zip](https://github.com/BiancoRoyal/node-red-contrib-iiot-opcua/files/3964013/ZeroASOFlowExample.json.zip)

closed 2019-12-14 by biancode