BiancoRoyal/node-red-contrib-iiot-opcua #115: Pass through from the "name" and "datatypeName" in the listener does not work

Flumpje, 2019-10-23:
**I'm submitting a ...**  (check one with "x")
   
## Feature
 - ( ) feature request
 
 **What is the motivation or Use Case for your feature request?**
 
 
 

## Support
 - support request => **Please do not submit support requests here**, instead see use [the link to our support page](https://bianco-royal.cloud/supporter/#/content/backer)
 - ( ) simple question => please visit the Wiki and Info tab by clicking on the node inside Node-RED before
 - FAQ: Can I send you individual support requests via E-Mail? 
   - Yes, sure! :-D
   
   
   

## Problem
 - ( X ) bug report => please search GitHub for a similar issue or PR before submitting
 - ( ) unexpected behaviour => please visit the Wiki before

### Information
 **Current behaviour**
 
Pass through from the "name" and "datatypeName" in the listener does not work
is: 
"addressSpaceItems": [{
        "name": "",
        "nodeId": "ns=2;s=Application.GVL_TTM.OpMode",
        "datatypeName": ""
}]
 
 **Expected behaviour**
 
Should be:
"addressSpaceItems": [{
        "name": "MachineStatus",
        "nodeId": "ns=2;s=Application.GVL_TTM.OpMode",
        "datatypeName": "Int16"
}] 

 **Minimal reproduction of the problem with instructions**
Inject nod -> Listener -> Debug

 
   1. Make data available in your opc server 
   2. 
   3. 
   

#### Your Environment
**Please tell us about your environment:**

  * ( ) I am using just the Node-RED package and got node-opcua installed with it
  * ( ) I have installed node-opcua global from source (using git clone)
  * ( X ) I have installed node-opcua as a global package (using npm i -g) 
  
  * Device :
    
    - ( ) macOS: v 
    - ( ) Linux: v 
    - ( X ) Windows: v 1803
    - ( ) Raspbian: v 
    - ( ) Other:
         - Name:
         - Version:
   
   * Description of the connecting OPC UA system:
   
     - Name   : Embedded control CML 75 | CPU: Intel Atom E3827 DualCore with 1.75 GHz  PLC.
     - Version : -
     - Vendor : Bosch Rexroth
     - Link   : http://www.boschrexroth.de
     
* **Node-RED version:** 

Node-RED version: v0.20.8

* **node-red-contrib-iiot-opcua version:** ``
 

* **Node:** (min. 8.x)
   `node --version` =
 v10.16.0

Flumpje, 2019-10-23:
[flows.zip](https://github.com/BiancoRoyal/node-red-contrib-iiot-opcua/files/3762362/flows.zip)

Teeffelen, 2019-11-18:
I think the issue may come from [this line](https://github.com/BiancoRoyal/node-red-contrib-iiot-opcua/blob/master/src/opcua-iiot-listener.js#L129) where the `name` and `datatypeName` are defined as empty, instead of using the values before the Listener node.

biancode, 2019-12-14:
@Teeffelen but there is no name and no datatype as you see ;)
in this version of node-opcua it is only to fix on browsing for every monitor event to read that information from the key (nodeId) - maybe I missed some, then let me know ...

biancode, 2019-12-14:
recipe to enhance:

* browse the information for NodeId on register monitored item
* save that information to a local dictionary
* fill up every monitor event with the dictionary information

biancode, 2019-12-15:
That could better and easier be done with a browse once for the node-id if it is unknown and organizing them in a context dictionary inside Node-RED.

closed 2019-12-15 by biancode