BiancoRoyal/node-red-contrib-iiot-opcua #120: **Address Space Script** of OPC UA server with Boolean, String or Datetime

dslgomes, 2020-01-11:
Hello, I trying use **Address Space Script** of OPC UA server with Boolean, String or Datetime but ins`t working. Could you help, what i need to do in code bellow to work?

	const gsamTopCyc = namespace.addVariable({
	  "organizedBy": samFolder,
	  "browseName": "TopCyc",
	  "nodeId": "ns=1;s=TopCyc",
	  "dataType": "**Boolean**",
		"value": {
		  "get": function() {
			return new Variant({
			  "dataType": DataType.**Boolean**,
			  //"value": enable
			  "value": flexServerInternals.sandboxFlowContext.get("samTopCyc")
			});
		  },
		  "set": function(variant) {
			flexServerInternals.sandboxFlowContext.set(
		"samTopCyc",
			  parseFloat(variant.value)
			);
			return opcua.StatusCodes.Good;
		  }
		}
		});

UlrichEntrenco, 2020-03-03:
Hi, I got the same issue. Is it a typo or is the datatype "boolean" not implemented yet?
Thanks!

biancode, 2020-03-04:
@UlrichEntrenco the type is implemented, but there is maybe a change and it does not work as expected anymore.

biancode, 2020-03-04:
please, try the new node-red-contrib-opcua-server

closed 2022-05-27 by biancode