BiancoRoyal/node-red-contrib-iiot-opcua #122: How to use setTimeout in address space script of OPC-UA-IIot-Flex-Server-Node?

JanSchmidtm, 2020-02-12:
Hello, 
I am trying to use a setTimeout-object in address space script of OPC-UA-IIoT-Flex-Server-Node but it isn`t working (see figure below). I get always a failure "Server Address Space Failure". Without the setTimeout-object everything works fine. How to implement timeouts in address space script?

Version of node-red-contrib-iiot-opcua: 3.12.0
Version of node-red: v1.0.3

[image]

Kind regards,

Jan

biancode, 2020-02-12:
My new package `node-red-contrib-opcua-server` has it.

JanSchmidtm, 2020-02-13:
Great, thanks for the information :-). 
It works!

closed 2022-05-27 by biancode