BiancoRoyal/node-red-contrib-iiot-opcua #126: Avoid too much data transfer

mhechthz, 2020-03-22:
Hi,

I have a question and I am not sure, wether this is the right place to ask. If not: sorry.

I have a simple workflow where I load several node configurations (finally >> 100) from a database like this:
```
var1, ns=3,s=Sawtooth, datatype=Float, name=myNode1
var1, ns=3,s=Square, datatype=Float, name=myNode2
...
```
I inject this to a listener node and get back something lie this:
`{"value":{"dataType":"Double","arrayType":"Scalar","value":1.8666666282348867},"statusCode":{"value":0,"description":"No Error","name":"Good"},"sourceTimestamp":"2020-03-22T09:37:07.000Z","sourcePicoseconds":0,"serverTimestamp":"2020-03-22T09:37:07.001Z","serverPicoseconds":0}`

Unfortunately there is no name of the submitted node, i.g. ' ns=3,s=Sawtooth'. If I now deselect "send only values", I get a really huge object back, including the crtificates, etc.

Would there be any chance to get also the names of the opcua objects in the "sparse" mode?

biancode, 2020-03-22:
@mhechthz not with that version, sorry. I do not get the information from node-opcua about that. I prepared the source for sending NodeIds, but it is not finished for now.

biancode, 2020-03-22:
Please, if you see a feature request here, then fill out the "new issue template" and send your issue with what is the use case and what is expected and a test case would be great.

closed 2020-03-22 by biancode

mhechthz, 2020-03-22:
I'm not sure, if I understand your answer. If I deactivate 'values only' I get the node id. Surprisingly node-red-contrib-opcua which is also based on node-opcua, gives the node id back in a very sparse object. But node-red-contrib-opcua is not very stable. That's why I switched to iiot-opcua.

biancode, 2020-03-22:
:) - have a look to the "package.json" files and see the difference I tried to explain:
node-red-contrib-opcua -> "node-opcua": "~2.4.4"
node-red-contrib-iiot-opcua ->  "node-opcua": "0.7.0"

biancode, 2020-03-22:
We need a good sponsor to upgrade to node-opcua 2.x to get a stable version again.

mhechthz, 2020-03-22:
Ah, ok. So I should request the update to version 2.4.4 or better ask for submission of node id?

biancode, 2020-03-22:
Version 2.x makes truly more sense here