BiancoRoyal/node-red-contrib-iiot-opcua #156: TypeError in OPCUA-IIoT-Write Node

AHyduchak, 2022-07-05:
### Which node-red-contrib-iiot-opcua version are you using?

4.0.1

### What happened?

I am unable to write a value to an OPC item. 

I have tried structuring the message where the `valuesToWrite` is `msg.valuesToWrite` and `msg.payload.valuesToWrite` as indicated in the **General Changes** of the description page.

The Write node returns:
`TypeError: nodeItem.indexOf is not a function`

### Server

Other/External server

### How can this be reproduced?

The problem can be reproduced with my flow as well the provided example flows
[flows.txt](https://github.com/BiancoRoyal/node-red-contrib-iiot-opcua/files/9048607/flows.txt)
.

AHyduchak, 2022-07-05:
[image]

The output of the debug node showing the msg contents.

ouinouin, 2022-07-10:
hi, same error here, i m trying to find more details on this, the error lies probably on the  : parseNamspaceFromItemNodeId function line 503 of the opcua-iiot-core.js , the function receives an item wich is a single value and tries to get the namespace from the nodeitem (wich contains the value) instead of the item.nodeId

i edited 
https://github.com/BiancoRoyal/node-red-contrib-iiot-opcua/blob/master/src/core/opcua-iiot-core.ts

(in fact locally i edited the equivalent javascript file (i dont know js enough to understand what are ts files )
line 513 
nodeNamespace = item.nodeId.substring(3, item.nodeId.indexOf(';'));

dtq-three, 2022-07-11:
This should be fixed with the changes from #157. 

> (in fact locally i edited the equivalent javascript file (i dont know js enough to understand what are ts files )

`.ts` files are [TypeScript](https://www.typescriptlang.org/), a superset of JavaScript that adds types. We compile those into JavaScript for distribution.

biancode, 2022-07-27:
E2E tests were disabled in TypeScript migration and is now active again see #177

closed 2022-10-12 by github-actions[bot]