BiancoRoyal/node-red-contrib-iiot-opcua #159: encodeObject Error when trying to perform Monitoring via a Listener Node

shantanoo-desai, 2022-07-11:
### Which node-red-contrib-iiot-opcua version are you using?

4.0.1

### What happened?

## Description

I am trying to perform a simple monitoring via the Listener Node on particular OPC-UA node. I have a simulated OPC-UA Server, via Node-RED v2.2.2 I am trying to observe the values from the OPC-UA node in the debug window.

### Environment

- Window 10 + Windows Subsystem for Linux 2 (Ubuntu 20.04)
- Docker / Docker Compose version (`v20.10.17/v2.6.0`)
- Prosys OPC UA Simulation Server (`v5.2.2-9`)
- Custom Node-RED Docker Container Image with `v2.2.2`
- `node-red-contrib-iiot-opcua` with `v4.0.1`

### Server

Other/External server

### How can this be reproduced?

## Reproduction Steps

### ProSys OPC UA Simulation Server

1. Install ProSys OPC UA Simulation Server
2. Under __Objects__ tab select `Simulation::FolderType` and figure out `NodeId` for `Counter` Simulator. In this case I have `NodeId` set to `n=3;i=1001`
3. Figure out the Connection Endpoint by clicking on `Status` tab in the UI

### Node-RED flow

Exported Flow to example (`endpoint` key value redacted, please adapt according to your endpoint via Step 3 above)

```json
[
    {
        "id": "f6f2187d.f17ca8",
        "type": "tab",
        "label": "Flow 1",
        "disabled": false,
        "info": ""
    },
    {
        "id": "98c6bdc0acd263b7",
        "type": "OPCUA-IIoT-Inject",
        "z": "f6f2187d.f17ca8",
        "injectType": "inject",
        "payload": "",
        "payloadType": "date",
        "topic": "",
        "repeat": "",
        "crontab": "",
        "once": false,
        "startDelay": "",
        "name": "",
        "addressSpaceItems": [
            {
                "name": "Counter",
                "nodeId": "ns=3;i=1001",
                "datatypeName": ""
            }
        ],
        "x": 320,
        "y": 180,
        "wires": [
            [
                "b112f2780d41b09f"
            ]
        ]
    },
    {
        "id": "448cbdb0f47b0a3e",
        "type": "OPCUA-IIoT-Node",
        "z": "f6f2187d.f17ca8",
        "injectType": "read",
        "nodeId": "ns=3;i=1001",
        "datatype": "Int32",
        "value": "",
        "name": "",
        "topic": "",
        "showErrors": true,
        "x": 330,
        "y": 260,
        "wires": [
            [
                "9b7775845c130e50"
            ]
        ]
    },
    {
        "id": "b112f2780d41b09f",
        "type": "OPCUA-IIoT-Listener",
        "z": "f6f2187d.f17ca8",
        "connector": "045fe0765df79790",
        "action": "subscribe",
        "queueSize": 10,
        "name": "",
        "topic": "",
        "justValue": false,
        "useGroupItems": false,
        "showStatusActivities": true,
        "showErrors": true,
        "x": 560,
        "y": 180,
        "wires": [
            [
                "8578c768c1f816d6"
            ]
        ]
    },
    {
        "id": "9b7775845c130e50",
        "type": "OPCUA-IIoT-Read",
        "z": "f6f2187d.f17ca8",
        "attributeId": 0,
        "maxAge": 1,
        "depth": 1,
        "connector": "045fe0765df79790",
        "name": "",
        "justValue": true,
        "showStatusActivities": false,
        "showErrors": true,
        "parseStrings": false,
        "historyDays": 1,
        "x": 570,
        "y": 260,
        "wires": [
            [
                "ef5b5c27896f94a1"
            ]
        ]
    },
    {
        "id": "8578c768c1f816d6",
        "type": "debug",
        "z": "f6f2187d.f17ca8",
        "name": "",
        "active": true,
        "tosidebar": true,
        "console": false,
        "tostatus": false,
        "complete": "true",
        "targetType": "full",
        "statusVal": "",
        "statusType": "auto",
        "x": 790,
        "y": 180,
        "wires": []
    },
    {
        "id": "ef5b5c27896f94a1",
        "type": "debug",
        "z": "f6f2187d.f17ca8",
        "name": "",
        "active": true,
        "tosidebar": true,
        "console": true,
        "tostatus": false,
        "complete": "payload.value[0].value",
        "targetType": "msg",
        "statusVal": "",
        "statusType": "auto",
        "x": 870,
        "y": 260,
        "wires": []
    },
    {
        "id": "ca99a87a46e8298e",
        "type": "inject",
        "z": "f6f2187d.f17ca8",
        "name": "",
        "props": [
            {
                "p": "payload"
            },
            {
                "p": "topic",
                "vt": "str"
            }
        ],
        "repeat": "",
        "crontab": "",
        "once": false,
        "onceDelay": 0.1,
        "topic": "",
        "payload": "",
        "payloadType": "date",
        "x": 160,
        "y": 260,
        "wires": [
            [
                "448cbdb0f47b0a3e"
            ]
        ]
    },
    {
        "id": "045fe0765df79790",
        "type": "OPCUA-IIoT-Connector",
        "discoveryUrl": "",
        "endpoint": "opc.tcp://<Redacted_HOSTNAME>:53530/OPCUA/SimulationServer",
        "keepSessionAlive": true,
        "loginEnabled": false,
        "securityPolicy": "None",
        "securityMode": "None",
        "name": "Prosys Simulation Server",
        "showErrors": false,
        "individualCerts": false,
        "publicCertificateFile": "",
        "privateKeyFile": "",
        "defaultSecureTokenLifetime": "",
        "endpointMustExist": false,
        "autoSelectRightEndpoint": false,
        "strategyMaxRetry": "",
        "strategyInitialDelay": "",
        "strategyMaxDelay": "",
        "strategyRandomisationFactor": "",
        "requestedSessionTimeout": "",
        "connectionStartDelay": "",
        "reconnectDelay": "",
        "maxBadSessionRequests": "10"
    }
]
```
[image]

### What did you expect to happen?

## Expectations

### What Works
Based on the flow the `Read` Node works perfectly well when trying to read the value of the `Counter` using `n=3;i=1001`

### What does NOT Work

Upon clicking the `timestamp` node connected to `Listener` node (which is configured to `n=3;i=1001`) I get the following error in the debug console of Node-RED

```javascript
Error: encodeObject Error: [Error] Value: {
  payload: {
    addressSpaceItems: [ [Object] ],
    nodetype: 'listen',
    injectType: 'subscribe',
    value: DataValue {
      statusCode: [ConstantStatusCode],
      sourceTimestamp: [Date],
      sourcePicoseconds: 0,
      serverTimestamp: [Date],
      serverPicoseconds: 0,
      value: [Variant]
    },
    monitoredItem: ClientMonitoredItemImpl {
      _events: [Object: null prototype],
      _eventsCount: 4,
      _maxListeners: undefined,
      statusCode: [ConstantStatusCode],
      subscription: [ClientSubscriptionImpl],
      itemToMonitor: [ReadValueId],
      monitoringParameters: [MonitoringParameters],
      monitoringMode: 2,
      timestampsToReturn: 2,
      result: [MonitoredItemCreateResult],
      monitoredItemId: 3,
      filterResult: undefined,
      [Symbol(kCapture)]: false
    }
  },
  topic: '',
  justValue: false,
  _msgid: 'ea1dd51655a1cc41'
}
```

This with the following `Listener` node configuration:

[image]

Upon selecting `Send Just Values` and restarting the modified flow, I am able to listen however I do not see any value returned

```json
{"payload":{"addressSpaceItems":[{"name":"","nodeId":"ns=3;i=1001","datatypeName":""}],"nodetype":"listen","injectType":"subscribe"},"topic":"","justValue":true,"_msgid":"4434a72fc40d5b99"}
```
Is this expected behaviour for the `Listener` node?

biancode, 2022-07-27:
see #160 - e2e tests of the Listener node had to be fixed soon and then we can check to fix your issue

shantanoo-desai, 2022-07-27:
@biancode thanks for reaching out. Let me know which part of the source code I can investigate to help out.

biancode, 2022-07-27:
Hi @shantanoo-desai the e2e tests and examples are the first points what we try to get working well again. The server code could be a good point to investigate at the moment via the e2e tests and examples

see [some instruction here](https://www.loom.com/share/ddf75a6c6d6345a9bbc5873670739467)

closed 2022-10-12 by github-actions[bot]