BiancoRoyal/node-red-contrib-iiot-opcua #190: Bug in core BuildVariant: Values like <0, 0.0, false> are resulting in creating null (or NaN) values

dei-p4i, 2022-08-04:
### Which node-red-contrib-iiot-opcua version are you using?

4.0.9

### What happened?

We encountered getting #NaN as values on our OPC UA Server, when using the write node, if the value was 0.0.
Further testing showed that also 0 (for int types) and false (for boolean) resulted in null errors.
Before the last update we used a 3.x.x version (don't remember exact version).

I'm probably found the problematic line of code in the source (see other information)

Pretty sure it's the if-test in the beginning of "buildNewVariant" in 
[/src/core/opcua-iiot-core.ts](https://github.com/BiancoRoyal/node-red-contrib-iiot-opcua/blob/master/src/core/opcua-iiot-core.ts):

```export function buildNewVariant(datatype: DataTypeInput, value: any): DataValueOptions {
  let variantValue: VariantOptions = {
    dataType: DataType.Null,
    value: null
  }
  // Possible bug location:
  // javascript interprets 0, 0.0 and false (or empty arrays or ...) here as false, so value stays "null"
  if (value.value)
    value = value.value```

### Server

OPCUA-IIoT-Flex-Server Node (Please attach AddressSpaceScript to the next section)

### How can this be reproduced?

Zipped Demo with Flex-Server using (slightly modified) original addressSpace code:
[demo-iiot-opcua-zero-value-bug.zip](https://github.com/BiancoRoyal/node-red-contrib-iiot-opcua/files/9259691/demo-iiot-opcua-zero-value-bug.zip)

- Set value to 1.0 by inject => read by inject => value is 1.0
- Set value to 0.0 by inject => read by inject => value is null

### What did you expect to happen?

Value should be 0.0 (or 0 or false depending on used data type)

biancode, 2022-08-07:
It would be good to have tests for this issue.

CW0815, 2022-08-20:
opc.tcp://milo.digitalpetri.com:62541/milo 
ns=2;s=CTT/Static/All Profiles/Scalar/Int16 

Write
Set value to 1 by inject (value is 1)
Set value to 0 by inject ("Error: expecting a number NaN")

Read
read 1 by inject (payload.value[0].value is 1)
read 0 by inject (payload.value[0].value.value is 0)

biancode, 2022-10-17:
We working on a version 4.1 - we found a lot issues after the TS merge, which are to be fixed soon.

CW0815, 2022-12-08:
same problem with version 4.1

opc.tcp://milo.digitalpetri.com:62541/milo
ns=2;s=CTT/Static/All Profiles/Scalar/Int16

Write
Set value to 1 by inject (value is 1)
Set value to 0 by inject ("Error: expecting a number NaN")
Set value to "0" by inject (value is 0)

Read
read 1 by inject (msg.payload.value[0].value is 1)
read 0 by inject (msg.payload.value[0].value.value is 0)

biancode, 2022-12-08:
We did not work on that for now. Maybe we have it next week in our next release.

biancode, 2022-12-09:
@JoelKrec we should include this as a test in the package.

biancode, 2022-12-09:
v4.1.1 actual test - release soon

[image]

biancode, 2022-12-09:
Test Flow to implement in e2e testings

[image]

[TestFlowZeroValueBug.json.zip](https://github.com/BiancoRoyal/node-red-contrib-iiot-opcua/files/10192573/TestFlowZeroValueBug.json.zip)

CW0815, 2022-12-14:
v4.1.1 the value is null, but with "OPCUA-IIoT-Inject" + "function" the value is 0

msg.payload.nodetype = 'inject';
msg.payload.injectType = 'write'

msg.payload.valuesToWrite = [];
msg.payload.valuesToWrite[0] = "0";

return msg;

[image]
[TestFlowZeroValueBug02.zip](https://github.com/BiancoRoyal/node-red-contrib-iiot-opcua/files/10231073/TestFlowZeroValueBug02.zip)

biancode, 2022-12-14:
@CW0815 give v4.1.2 a try

CW0815, 2022-12-15:
This works with v4.1.2

But with a response between Read and helper,
the output is different at 0 and 1.

[image]

biancode, 2022-12-16:
The response node is not working very well after the shift to TypeScript. See the [TODO.md](https://github.com/BiancoRoyal/node-red-contrib-iiot-opcua/blob/master/TODO.md) for more information.

closed 2023-03-02 by github-actions[bot]