BiancoRoyal/node-red-contrib-iiot-opcua #202: OPCUA-IIoT-Node does not handle Uint data types correctly

patstave, 2022-10-12:
### Which node-red-contrib-iiot-opcua version are you using?

4.0.11

### What happened?

It seems to me that the OPCUA-IIoT-Node is not able to handle UInt16 and UInt32 data types.

The code:
`[
    {
        "id": "f110299a341b5a59",
        "type": "debug",
        "z": "0f0e227c150104a3",
        "name": "debug",
        "active": true,
        "tosidebar": true,
        "console": false,
        "tostatus": false,
        "complete": "true",
        "targetType": "full",
        "statusVal": "",
        "statusType": "auto",
        "x": 2330,
        "y": 1240,
        "wires": []
    },
    {
        "id": "f57352fbd175b6a3",
        "type": "OPCUA-IIoT-Node",
        "z": "0f0e227c150104a3",
        "injectType": "write",
        "nodeId": "ns=2;s=Tunga_Mitsubishi.ElolinkPLC.TM20_Anritsu_Inline.TM20InlineCounter",
        "datatype": "UInt16",
        "value": "39",
        "name": "Node",
        "topic": "",
        "showErrors": true,
        "x": 2170,
        "y": 1240,
        "wires": [
            [
                "f110299a341b5a59",
                "5c1540a1d4a9c293"
            ]
        ]
    },
    {
        "id": "9b85a0bc5b1d94b6",
        "type": "inject",
        "z": "0f0e227c150104a3",
        "name": "",
        "props": [
            {
                "p": "payload.timestamp",
                "v": "",
                "vt": "date"
            },
            {
                "p": "topic",
                "vt": "str"
            }
        ],
        "repeat": "",
        "crontab": "",
        "once": false,
        "onceDelay": 0.1,
        "topic": "",
        "x": 2030,
        "y": 1240,
        "wires": [
            [
                "f57352fbd175b6a3"
            ]
        ]
    }
]`

...produces the following message object:
`object
payload: object
nodetype: "node"
injectType: "write"
addressSpaceItems: array[1]
0: object
name: "Abc"
nodeId: "ns=2;s=Tunga_Mitsubishi.ElolinkPLC.TM20_Anritsu_Inline.TM20InlineCounter"
datatypeName: "UInt16"
valuesToWrite: array[1]
0: 0
value: undefined
topic: ""
_msgid: "31786be882dd5f2b"`

The value of `msg.payload.addressSpaceItems.ValuesToWrite[0]` is set to a value of "0". It should be "39" according to the code above.

If i change  `"datatype": "UInt16"` to `"datatype": "Int16"` the value of `msg.payload.addressSpaceItems.ValuesToWrite[0]` is set correctly.

The decimal value of "39" is a valid UInt16 and also a valid Int16. Why does the output of OPCUA-IIoT-Node return "0" when set to Uint16?

### Server

None/This is related to a node that doesn't connect to a server

### How can this be reproduced?

`[
    {
        "id": "f110299a341b5a59",
        "type": "debug",
        "z": "0f0e227c150104a3",
        "name": "debug",
        "active": true,
        "tosidebar": true,
        "console": false,
        "tostatus": false,
        "complete": "true",
        "targetType": "full",
        "statusVal": "",
        "statusType": "auto",
        "x": 2330,
        "y": 1240,
        "wires": []
    },
    {
        "id": "f57352fbd175b6a3",
        "type": "OPCUA-IIoT-Node",
        "z": "0f0e227c150104a3",
        "injectType": "write",
        "nodeId": "ns=2;s=Tunga_Mitsubishi.ElolinkPLC.TM20_Anritsu_Inline.TM20InlineCounter",
        "datatype": "UInt16",
        "value": "39",
        "name": "Node",
        "topic": "",
        "showErrors": true,
        "x": 2170,
        "y": 1240,
        "wires": [
            [
                "f110299a341b5a59",
                "5c1540a1d4a9c293"
            ]
        ]
    },
    {
        "id": "9b85a0bc5b1d94b6",
        "type": "inject",
        "z": "0f0e227c150104a3",
        "name": "",
        "props": [
            {
                "p": "payload.timestamp",
                "v": "",
                "vt": "date"
            },
            {
                "p": "topic",
                "vt": "str"
            }
        ],
        "repeat": "",
        "crontab": "",
        "once": false,
        "onceDelay": 0.1,
        "topic": "",
        "x": 2030,
        "y": 1240,
        "wires": [
            [
                "f57352fbd175b6a3"
            ]
        ]
    }
]`

### What did you expect to happen?

Expected value of `msg.payload.addressSpaceItems.ValuesToWrite[0]`: 39
Actual value of `msg.payload.addressSpaceItems.ValuesToWrite[0]`: 0

### Other Information

Node-RED v. 3.0.2
Node.JS v. 18

biancode, 2022-10-17:
We working on a version 4.1 - we found a lot issues after the TS merge, which are to be fixed soon.

patstave, 2022-11-15:
Is there a timeline for an estimated release of 4.1?

Should i work with 3.12.1 in the meantime?

biancode, 2022-12-09:
should work from v4.1.1+

patstave, 2022-12-20:
I can confirm that this has been resolved in V4.1.2

closed 2022-12-20 by patstave