BiancoRoyal/node-red-contrib-iiot-opcua #219: Listener - Output an array

dpatel2016, 2023-02-23:
### Which node-red-contrib-iiot-opcua version are you using?

4.1.2

### What happened?

I have about 50 variables I am trying to read. I am using a listener to update the variables within the flow, but I would like to get an array of all 50 variables if any variable has a change. Currently, it only outputs the node that has a change. 

### Server

Other/External server

### How can this be reproduced?

1. add Inject node 
2. add listen node 
3. add debug node
4. configure them all to read tags from the server.

biancode, 2023-02-23:
The listen mode is the monitored item of OPC UA. If you need such a function, then it has to be done in your flow. It is not a bug or part of the package at the moment.

Our new PLUS for Node-RED OPC UA nodes will provide such features, soon.

dpatel2016, 2023-02-23:
That makes sense, thank you. 
Any suggestion on how to do it in the flow?

biancode, 2023-02-23:
@dpatel2016 you can use the flow context to store all changes in an easy to use array or dictionary structure and if it has a changed event, then you can send out all data after the change of your context model right out of the context model see [Working with the Node-RED Context](https://nodered.org/docs/user-guide/context)

dpatel2016, 2023-02-23:
@biancode that's what I was doing with the read node initially. The benefit I had with the read node's output of an array was that I can connect one function node to the read node and set up all context variables using array index such as msg.payload.value[32]. 

With the current output from the listener node, the path to the value is very vague such as msg.payload.value.value.value. Now I have to all conditions to make sure the correct context variable gets updated.

biancode, 2023-02-23:
I'm sorry for that state. We are working since last year September to provide a very new OPC UA packages for Node-RED to make all those points much easier soon, but it takes some time. Watch for the plus4nodered packages inside Node-RED in April!

dpatel2016, 2023-02-23:
@biancode No worries. The current features are great and very helpful but I wanted to see if there was an easier way than what I had in mind. 
Thank you for creating and supporting this library

closed 2023-02-23 by dpatel2016